argoverse/av2-api #7: 3D maps but 2D tracks

ZikangZhou, 2022-03-18:
It seems that the map elements have 3D coordinates but the tracks are still in 2D. Why the dataset does not provide 3D tracks? Is it possible to infer the z coordinate of the tracks?

Another quick question: will the future release includes the traffic light state?

wqi, 2022-03-20:
Hi Zikang, thanks for your interest in the motion forecasting dataset!

To provide a bit more context: the Argoverse 2 Motion Forecasting Dataset, like the Argoverse 1 Motion Forecasting Dataset, is focused on 2D (bird's eye view) forecasting of objects. 

We don't expect there to be much interesting behavior in the Z direction. Objects generally stay on the ground. We provide maps in 3D because the map format is shared between all Argoverse datasets, but you can project them to 2D trivially by removing the z coordinate. 

We do plan to include dynamic map information (such as traffic light state) in a future release of the MF dataset, so please keep your eyes peeled! We will announce future updates through the website and through release notes on this repo.

ZikangZhou, 2022-03-20:
Thank you very much! I can’t wait to try the new dataset.

closed 2022-03-20 by ZikangZhou