argoverse/av2-api #11: Is it possible to extract the route information ?

panda2020-sky, 2022-03-20:
Hi, thank you for providing the outstanding dataset.

I am particularly interested in the motion dataset, and have a question that is it possible to extract the route of the self-driving vehicles in each scenario?

wqi, 2022-03-21:
Hello! Thanks for your interest in the AV2 MF dataset.

Yes, you should definitely be able to extract the route of the SDV in each scenario.

You can do this by extracting the location history of the track with `track_id="AV"` in each scenario, which is guaranteed to exist.

panda2020-sky, 2022-03-22:
> Hello! Thanks for your interest in the AV2 MF dataset.
> 
> Yes, you should definitely be able to extract the route of the SDV in each scenario.
> 
> You can do this by extracting the location history of the track with `track_id="AV"` in each scenario, which is guaranteed to exist.

Thank you for your answer. And sorry for not expressing clearly in my question described above.
The 'route' in my question represents the planning trajectory inputted to the AV, instead of the actual historical trajectory of the AV.
I think this information can be obtained by the planning module of the AV, or from the map API. But I don't know how to do it...

wqi, 2022-03-22:
Unfortunately we don't provide information related to the AV's routing/planning as part of the current MF dataset.

However, this is something we're certainly exploring for future iterations of Argoverse! :-)

panda2020-sky, 2022-03-22:
Looking forward!

closed 2022-03-22 by panda2020-sky