argoverse/av2-api #12: Download the trajectory prediction part of the map dataset

12num, 2022-03-21:
Hello, we have a question that needs to be interrupted. The direction of our research is trajectory prediction. Your description of the map dataset is that it contains four maps of trajectory prediction, sensor, radar, and Tbv. We wonder if we only need the part about 250,000 vector graphics for trajectory prediction, the official download link has 21 parts in the map dataset itself. Can you tell us which part is the trajectory prediction part? We would appreciate it if you could.

johnlambert-argo, 2022-03-21:
Hi @12num, thanks for your interest in Argoverse 2. Argoverse 2 consists of four completely independent datasets. Each of the 4 datasets comes with their own paired local maps for each scenario (this is different from Argoverse 1, so I understand the confusion). Please take a look at this section of the [readme](https://github.com/argoai/argoverse2-api#datasets) -- we recently added this table to clarify this:

[image]

If you're interested in the trajectory prediction task, and not the map change detection task, please take a look at the [Motion Forecasting Dataset Readme](https://github.com/argoai/argoverse2-api/blob/main/src/av2/datasets/motion_forecasting/README.md) and the [Argoverse 2 Download Page](https://www.argoverse.org/av2.html#download-link):

[image]
You'll want to use the links in the bottom-left quadrant.

Alternatively, you can use `s5cmd` to pull down the motion forecasting data, as described [here](https://github.com/argoai/argoverse2-api/blob/main/DOWNLOAD.md#dataset-s3-locations).

12num, 2022-03-22:
Thanks for your replies, it is very helpful for me

closed 2022-03-22 by 12num