argoverse/av2-api #17: Downloading datasets using S3

neer201, 2022-03-23:
Hello, there! I tried to download av2 datasets using S3, I rely on these [instructions](https://github.com/argoai/av2-api/blob/main/DOWNLOAD.md).

But it doesn't work. I try both: s5cmd and usual awscli clients.

1) s5cmd:
```bash
s5cmd --no-sign-request cp s3://argoai-argoverse/av2 .

ERROR "cp s3://argoai-argoverse/av2 av2/av2": AccessDenied: Access Denied status code: 403, request id: 2HZ44EQ5VX276MJ5, host id: 8NIIVQuhziyBavM4B98jmDrhj6Qpbyw4FbTFo76M6wtc3RHhQRehm7I83ZspMdOjThNy1RUv0Io=
```

2) aws cli
```bash
aws s3 ls s3://argoai-argoverse/av2

An error occurred (AccessDenied) when calling the ListObjectsV2 operation: Access Denied
```

benjaminrwilson, 2022-03-23:
Hi @neer201, thanks for your interest.

Would you mind trying:

```
s5cmd --no-sign-request cp s3://argoai-argoverse/av2/* av2
```

Let me know if this helps!

neer201, 2022-03-23:
> Hi @neer201, thanks for your interest.
> 
> Would you mind trying:
> 
> ```
> s5cmd --no-sign-request cp s3://argoai-argoverse/av2/* av2
> ```
> 
> Let me know if this helps!

It works, thanks a lot!

closed 2022-03-23 by neer201