argoverse/av2-api #18: Will Argoverse V2 provide the 2D bbox (length & width) of objects?

yenw, 2022-03-23:
Argoverse V1.1 & V2 is missing the 2D bbox (length & width) of objects.
Will Argoverse V2 provide these data?

johnwlambert, 2022-03-23:
Hi @yenw, thanks for your interest in Argoverse 2.

In the Argoverse v1.1 and v1.2 **3D Tracking Dataset**, each object cuboid does come with a length and width. For the **Motion Forecasting** datasets in v1.1 and v1.2, we didn't provide an object size.

Our Argoverse 2 release consists of 4 datasets -- a **Sensor Dataset** with full sensor suite measurements and labeled cuboids (1K logs w/ object length, width, height), a **Motion Forecasting Dataset** with object tracks and categories (250K logs), a large unlabeled **LiDAR Dataset** (20K logs), and a **Map Change Dataset** (1000+ logs).

Could you tell us a bit more about which datasets you are referring to and are hoping to work with?

yenw, 2022-03-24:
> Hi @yenw, thanks for your interest in Argoverse 2.
>
> In the Argoverse v1.1 and v1.2 **3D Tracking Dataset**, each object cuboid does come with a length and width. For the **Motion Forecasting** [...]

Hi @johnwlambert, thanks for your reply.
I hope ArgoverseV2 can provide the length and width of object in `Motion Forecasting Dataset`. The length and width are important features for `Motion Forecasting`. In addition to this, other datasets provide the size of objects, such as `Waymo Motion Dataset` and `Shifts Vehicle Motion Prediction dataset`.

wqi, 2022-04-07:
Hi @yenw, unfortunately we don't provide those attributes for actors in the motion forecasting dataset at this time.

However, we've noted your interest in these attributes and will certainly consider adding them in alongside a future release. :-)

closed 2022-04-07 by wqi

James-Hays, 2022-04-08:
@yenw I will point out that unlike the Argoverse 1 Motion Forecasting Dataset, the Argoverse 2 Motion Forecasting Dataset provides the object class for each actor. So you know if it's a bicycle, or a bus, or a passenger vehicle, etc. This gives a strong prior on the spatial extent of the object.