argoverse/av2-api #34: map_tutorial.ipynb

tom-bu, 2022-04-03:
In your map_tutorial.ipynb, I encountered some issues: 
`from av2.datasets.sensor_dataset.av2_sensor_dataloader import AV2SensorDataLoader`
should be 
`from av2.datasets.sensor.av2_sensor_dataloader import AV2SensorDataLoader`

Each instance of ` ArgoverseStaticMap.from_json()` should be ` ArgoverseStaticMap.from_map_dir()`

And then `args` should be like

`from pathlib import Path`
`args = SimpleNamespace(**{"dataroot": Path(dataroot), "log_id": Path(log_id)})`

benjaminrwilson, 2022-04-03:
Thanks for pointing this out, @tom-bu. Would you mind checking to see if this PR fixes the issues for you? https://github.com/argoai/av2-api/pull/35

tom-bu, 2022-04-04:
Yes, I believe it's fixed now.

closed 2022-04-04 by tom-bu