argoverse/av2-api #36: ValueError: 'UNKNOWN' is not a valid LaneMarkType

kingwmk, 2022-04-07:
Hello, Thanks for your great contributions.
I found an error when parsing map data. 
For example for log_map_archive_f75d80d6-a1c8-4a0b-915a-7dc304d122b0.json:

Traceback (most recent call last):
  File "map_test.py", line 28, in <module>
    avm = ArgoverseStaticMap.from_json(Path(dataroot))
  File "/Users/wangmingkun/opt/anaconda3/envs/argo2/lib/python3.8/site-packages/av2/map/map_api.py", line 328, in from_json
    vector_lane_segments = {ls["id"]: LaneSegment.from_dict(ls) for ls in vector_data["lane_segments"].values()}
  File "/Users/wangmingkun/opt/anaconda3/envs/argo2/lib/python3.8/site-packages/av2/map/map_api.py", line 328, in <dictcomp>
    vector_lane_segments = {ls["id"]: LaneSegment.from_dict(ls) for ls in vector_data["lane_segments"].values()}
  File "/Users/wangmingkun/opt/anaconda3/envs/argo2/lib/python3.8/site-packages/av2/map/lane_segment.py", line 110, in from_dict
    left_mark_type=LaneMarkType(json_data["left_lane_mark_type"]),
  File "/Users/wangmingkun/opt/anaconda3/envs/argo2/lib/python3.8/enum.py", line 339, in __call__
    return cls.__new__(cls, value)
  File "/Users/wangmingkun/opt/anaconda3/envs/argo2/lib/python3.8/enum.py", line 663, in __new__
    raise ve_exc
ValueError: 'UNKNOWN' is not a valid LaneMarkType

kingwmk, 2022-04-08:
I add a "UNKNOWN" type for LaneMarkType. It seems work. 
class LaneMarkType(str, Enum):
    """Color and pattern of a painted lane marking, located on either the left or ride side of a lane segment.
    The `NONE` type indicates that lane boundary is not marked by any paint; its extent should be implicitly inferred.
    """

    DASH_SOLID_YELLOW: str = "DASH_SOLID_YELLOW"
    DASH_SOLID_WHITE: str = "DASH_SOLID_WHITE"
    DASHED_WHITE: str = "DASHED_WHITE"
    DASHED_YELLOW: str = "DASHED_YELLOW"
    DOUBLE_SOLID_YELLOW: str = "DOUBLE_SOLID_YELLOW"
    DOUBLE_SOLID_WHITE: str = "DOUBLE_SOLID_WHITE"
    DOUBLE_DASH_YELLOW: str = "DOUBLE_DASH_YELLOW"
    DOUBLE_DASH_WHITE: str = "DOUBLE_DASH_WHITE"
    SOLID_YELLOW: str = "SOLID_YELLOW"
    SOLID_WHITE: str = "SOLID_WHITE"
    SOLID_DASH_WHITE: str = "SOLID_DASH_WHITE"
    SOLID_DASH_YELLOW: str = "SOLID_DASH_YELLOW"
    SOLID_BLUE: str = "SOLID_BLUE"
    NONE: str = "NONE"
    **_UNKNOWN: str = "UNKNOWN"_**

I am not sure if is this in your data definition.

wqi, 2022-05-13:
Hi @kingwmk, thanks for reporting this bug! We've landed #58 to address this issue.

closed 2022-05-13 by wqi