argoverse/av2-api #37: Cuboids to BBoxes

sshefs02, 2022-04-07:
Hi! 

I was going through Argoverse 1 dataset and found this file: https://github.com/argoai/argoverse-api/blob/7c4952d4d080b5b173c25e187c46d3293245a67b/demo_usage/cuboids_to_bboxes.py#L105 that can render 3D bounding boxes on images as per README. Is there a similar function call in the av2 APIs as well? Alternatively, is there a documentation around how to do this? 

Thank you!

benjaminrwilson, 2022-04-13:
Hi @sshefs02,

>  Is there a similar function call in the av2 APIs as well?

Here is a function that projects cuboids into the viewing frustum of _one_ ring camera: https://github.com/argoai/av2-api/blob/15d43e3699bb5626009000847b877e8053dd03c0/src/av2/structures/cuboid.py#L276.

> Alternatively, is there a documentation around how to do this?

The tutorial is being finalized (some aspects may be tweaked before merging into `main`), but this may be helpful to you: https://github.com/benjaminrwilson/av2-api/blob/feature/sensor-viz/tutorials/generate_sensor_dataset_visualizations.py.

Let me know if you run into any issues!

benjaminrwilson, 2022-05-05:
Closing due to inactivity. Feel free to reach out if you have any other questions!

closed 2022-05-05 by benjaminrwilson