argoverse/av2-api #38: How to evaluate 3D object detection on validation split?

Abyssaledge, 2022-04-09:
Thanks for your excellent work!
I would like to know how to do the evaluation of 3D object detection on validation split. 
And I notice there is PR about this. When will the stable version be released? I am looking forward to it!

benjaminrwilson, 2022-04-11:
Hi @Abyssaledge, the detection evaluation has been merged into `main`. We will tag a versioned release later this week with the final configuration. If you'd like a starting point with an example of how to use the evaluation code, you can check out the 3D object detection baseline found here: https://github.com/benjaminrwilson/torchbox3d.

Let me know if you run into any issues.

Abyssaledge, 2022-04-12:
Thanks very much. I am going to have a try. I hope to keep this issue open in the next couple of days for further discussion and community reference.

benjaminrwilson, 2022-05-10:
Hi @Abyssaledge, our 2022 3D object detection challenge is now live! Please have a look when you get the chance: https://eval.ai/challenge/1710/overview.

benjaminrwilson, 2022-05-12:
Let me know if you have any questions!

closed 2022-05-12 by benjaminrwilson