argoverse/av2-api #46: Argoverse 2.0 vs Argoverse 1.1 API

Cram3r95, 2022-04-27:
Hi folks,

I am trying to run my model in Argoverse 2.0, which was previously trained using 1.1 and its corresponding API. Nevertheless, after installing and cloning the API, in order to check the tutorials, dataloaders etc, this api looks quite smaller than Argoverse 1.1. and the organization also seems to be different (e.g. where are the csvs with the trajectories?). Where could I see all the required documentation?

jagjeet-singh, 2022-04-27:
Hi,

Thanks for your continued interest in Argoverse!

Argoverse 2.0 API does not inherit from Argoverse 1.1 API. Since the underlying data (and its representation) is fundamentally different, it made sense to create a new API from (almost) scratch. Adapting Argoverse 1.1 baselines on 2.0 was not straightforward, and we plan to release the code soon.

Until then, you can start from this [README](https://github.com/argoai/av2-api/blob/main/src/av2/datasets/motion_forecasting/README.md). There are also some utils [here](https://github.com/argoai/av2-api/blob/15d43e3699bb5626009000847b877e8053dd03c0/tutorials/generate_forecasting_scenario_visualizations.py#L66) that demonstrate how to work with forecasting data (parquet) and map data (json).

Hope this helps!

Cram3r95, 2022-04-27:
Then, how is it possible to participate in the Workshop of Autonomous
Driving (CVPR) without this code? Or is It Argoverse 1.1. based?

jagjeet-singh, 2022-04-27:
The API currently has everything to start training models on Argoverse 2.0. The baseline code will just provide another reference on how to use 2.0 API, so it is an optional feature, not a blocker.

We do plan to provide utils for participating in CVPR challenge (which is independent of the API). Stay tuned!

panda2020-sky, 2022-04-27:
> The API currently has everything to start training models on Argoverse 2.0. The baseline code will just provide another reference on how to use 2.0 API, so it is an optional feature, not a blocker. [...]

Hi, is there any API in argoverse 2 that is similar to the' ArgoverseForecastingLoader' in argoverse1?
It seems that no data_loading module exists in argo2 that can be used to conveniently load the data.

jagjeet-singh, 2022-04-27:
We don't have such a dataloader for Argoverse 2.0 but we have included all the necessary utils [here](https://github.com/argoai/av2-api/blob/main/src/av2/datasets/motion_forecasting/scenario_serialization.py). We have found that people often built their own data loaders which aligns better with their models.

If you think a generic dataloader can still be helpful, please feel free to create a feature request or a PR that we can review. Thanks!

closed 2022-05-06 by wqi

Cram3r95, 2023-02-13:
Hi, regarding the problem with Argo AI, Ford and Volkswagen, is the Argoverse 2.0 (specially Motion Forecasting) be mantained over time?

wqi, 2023-02-13:
Hi Carlos, custody of the AV2 dataset has been transferred over to Georgia Tech.

The core team remains committed to the long term maintenance and stewardship of existing data, under the same terms used previously.

Cram3r95, 2023-02-14:
Thank you so much @wqi, since nobody is answering the questions (specially those related with map in Motion Forecasting), I was afraid of this. Do you know how to generate a local map and plausible centerlines around the focal agent in a similar way to Argoverse 1.0?

Here are the 1.0 functions:

https://github.com/argoverse/argoverse-api/blob/f886ac54fba9f06f8a7d109eb663c7f501b3aa8e/argoverse/map_representation/map_api.py#L830