argoverse/av2-api #55: Baseline models

Cram3r95, 2022-05-08:
Hi guys,

Do you know if some GNN based baseline methods are going to be implemented in the tutorial? Similar than NN and LSTM for Argoverse 1.0 API, in order to know how to deal with the vectorized map.

wqi, 2022-05-13:
Hi @Cram3r95, thanks for your interest in the AV2 MF dataset! It is in the cards to update the [WIMP](https://github.com/wqi/WIMP) baseline with a preprocessing script compatible with AV2, but we don't have an ETA quite yet.

We'll be sure to update this issue when the baseline is ready.

Cram3r95, 2022-05-13:
Hi @wqi, that would be fantastic, since AV2 presents more information in order to leverage the vectorized hdmap, so GNNs could be a good option. Thank you!

closed 2022-05-13 by Cram3r95