argoverse/av2-api #60: When will the Argoverse 2: 3D Object Detection Competition ends ?

Icecream-blue-sky, 2022-05-17:
I would like to confirm if the competition ends on June 13 ？

James-Hays, 2022-05-17:
Hi,

Yes, all competitions ([Stereo](https://eval.ai/challenge/1704/overview), [Motion Forecasting](https://eval.ai/challenge/1719/overview), [3D Object Detection](https://eval.ai/challenge/1710/overview)) end on 11:59PM EST on June 13th, 2022. The leaderboards will stay online, though.

closed 2022-05-18 by Icecream-blue-sky

Icecream-blue-sky, 2022-05-20:
Two friendly advice: 1、The website(https://cvpr2022.wad.vision/) still shows that the game has not started, which I think will mislead some people. 
[image]
2、Why not publish the contest in codalab? I find it difficult to find the entrance to this contest (at least not as easy as codalab), I think that's why this contest has a low number of entries.