argoverse/av2-api #66: 'UNKNOWN' is not a valid LaneMarkType in v0.2.0

godture, 2022-05-27:
Hi,
I get the following error running code `ArgoverseStaticMap.from_json(static_map_path)` with certain map files:
`ValueError: 'UNKNOWN' is not a valid LaneMarkType`
The version of av2 is 0.2.0.
When I check the LaneMarkType class while running, I notice that it does not have the member "UNKNOWN", which exists in the github code.
Could you please check and fix it in the released version?

karthi0804, 2022-05-30:
Please let me know if there is a quick fix.

wqi, 2022-06-02:
Hi folks, we just released AV2 v0.2.1 to PyPI that includes the fix for this issue - please try updating your API package.

If you're still experiencing issues, please feel free to re-open this issue. :-)

closed 2022-06-02 by wqi