argoverse/av2-api #68: About Motion Forecasting Challenge Submission

mikezhang95, 2022-05-28:
Hi! I am about to make a submission to the motion forecasting challenge (https://eval.ai/challenge/1719/my-submissions) but got an error "KeyError(1)" from Stderr File. Is there any way to get a detailed error message?

chch9907, 2022-05-30:
I also got this issue and I found it was because my predictions did not include all the scenarios in test data, which might result from the data preprocessing.  Actually, scenario id is the key of the dict you submit. Hope it could help you.

mikezhang95, 2022-06-01:
> I also got this issue and I found it was because my predictions did not include all the scenarios in test data, which might result from the data preprocessing. Actually, scenario id is the key of the dict you submit. Hope it could help you.

Thanks for replying. Your suggestion solves my problem. I can submit now.

closed 2022-06-01 by mikezhang95