argoverse/av2-api #84: Questions about the height (z axis) of 3D lane

qiaozhijian, 2022-08-01:
I found that the height of 3d lanes might be a little buggy. It is somewhat discontinuous and appears inaccurate. You can see the example as follows. Different views are presented.

[image]

However, if I project them to 2D space by removing z axis, they will look good.

[image]

Could you please tell me how the height is obtained? By lidar?

How can I process the lanes? Only use XY?

James-Hays, 2022-08-02:
Hi @qiaozhijian 

It's definitely possible that there are issues with the data, but it looks like the axes you are using to visualize the data are vastly different scales. The x and y dimensions span 60 or 70 meters but the height dimension spans only 60 centimeters. So maybe that is correct and this scene is fairly flat?

qiaozhijian, 2022-08-05:
> 
Indeed, each axis has a different scale due to the default settings of matplotlib. But let's take a closer look at the absolute scale, the lanes have a height change of nearly 1m in a local area, which is obviously unreasonable. Of course, if the scale is uniform, then the scene will look flat.

qiaozhijian, 2022-08-05:
> Hi @qiaozhijian
>
> It's definitely possible that there are issues with the data, but it looks like the axes you are using to visualize the data are vastly different scales. The x and y dimensions span [...]

Thank your for your reply. Indeed, each axis has a different scale due to the default settings of matplotlib. But let's take a closer look at the absolute scale, the lanes have a height change of nearly 1m in a local area, which is obviously unreasonable. Of course, if the scale is uniform, then the scene will look flat.

qiaozhijian, 2022-08-05:
I noticed that av2 also provided rasterized map of ground height, I thought it would be better to use that to correct the lanes.

qiaozhijian, 2022-08-09:
Hi @benjaminrwilson, I'm glad to provide the whole visualization code.
```
from pathlib import Path
import matplotlib.pyplot as plt
import numpy as np
from av2.datasets.sensor.av2_sensor_dataloader import AV2SensorDataLoader
from av2.map.map_api import ArgoverseStaticMap, LaneSegment

if __name__ == '__main__':
    # path to where the logs live
    dataroot = Path("/media/qzj/Extreme SSD/datasets/argoverse/tbv/logs/")
    # unique log identifier
    log_id = "2S1xkgWyVhoa5zwPLz9tl2MAG0lMxj6i__Autumn_2020"

    loader = AV2SensorDataLoader(data_dir=dataroot, labels_dir=dataroot)
    log_map_dirpath = Path(dataroot) / log_id / "map"
    avm = ArgoverseStaticMap.from_map_dir(log_map_dirpath, build_raster=False)

    # retain every pose first.
    traj_ns = loader.get_subsampled_ego_trajectory(log_id, sample_rate_hz=1e9)
    med_x, med_y = np.median(traj_ns, axis=0) # example 1
    # med_x, med_y = traj_ns[10]    # example 2

    # Derive plot area from trajectory (with radius defined in infinity norm).
    view_radius_m = 20
    xlims = [med_x - view_radius_m, med_x + view_radius_m]
    ylims = [med_y - view_radius_m, med_y + view_radius_m]

    fig = plt.figure()
    ax = fig.add_subplot(111, projection='3d')
    vector_lane_segments = avm.get_nearby_lane_segments(query_center=np.asarray([med_x, med_y]), search_radius_m=view_radius_m)
    for vls in vector_lane_segments:
        left_lane_markings = vls.left_lane_marking.polyline
        right_lane_markings = vls.right_lane_marking.polyline
        ax.plot(left_lane_markings[:, 0], left_lane_markings[:, 1], left_lane_markings[:, 2], linewidth=1)
        ax.plot(right_lane_markings[:, 0], right_lane_markings[:, 1], right_lane_markings[:, 2], linewidth=1)

    plt.xlim(*xlims)
    plt.ylim(*ylims)
    plt.legend()
    plt.tight_layout()
    plt.show()
```