argoverse/av2-api #87: Similarity argoverse 1 / argoverse 2

odunkel, 2022-08-23:
Hey,
the argoverse 2 dataset comes with new and richer scenes. Comparing the scenes of av1 to av2 in the respective cities: How similar what you consider them? So, in short: Would you say training with argoverse 2 includes all the relevant data to perform well on argoverse 1? I would be particularly interested in the motion forecasting dataset.
Looking forward to your answer! 
Thanks a lot!

James-Hays, 2022-08-24:
Hi @odunkel 

I think this depends on the particular task you are working on. For lidar-based tasks, the datasets look pretty similar. AV1 and AV2 use the same lidar sensor. For image-based tasks, we sort of expect things to generalize pretty well across cameras. The cameras are different, though. If your task is something like automatically predicting the map, then that might be more of an issue because the underlying map representation changed. 

In general, AV2 scenarios contain a superset of information contained in AV1 scenarios. AV2 scenarios are also denser and more diverse. So yeah, big picture, I'd expect something trained on AV2 to perform well on AV1, assuming that comparison is even possible. E.g. you can't test your AV2 wheelchair detector on AV1, because wheelchairs are not in the AV1 taxonomy.

odunkel, 2022-08-25:
Hey, thanks for your answers. Yeah, I've already looked into the figures in the paper and they are definiteley helpful. My question was rather specifically directed to whether av2 "includes" all the scenario types av1 includes. For example, does av2 include the same / very similar roadways, set of surrounding agents and interactions, etc. av1 includes?
But, I guess, your comment, @James-Hays, answered that question. Thanks!

wqi, 2022-09-20:
Hi @odunkel, apologies for the delayed response. The AV2 MF dataset could be viewed as a "superset" of the AV1 MF dataset when viewed from the perspective of roadway, actor, and interaction types. 

We'd expect a model trained on AV2 data to perform better on AV1 data than the other way around, due to the addition of additional actor types and cities compared to the previous iteration of Argoverse.

If you have any additional questions relating to the composition of AV2, please feel free to re-open this issue! :-)

closed 2022-09-20 by wqi