argoverse/av2-api #92: lane label annotation mathod inquery.

Mollylulu, 2022-09-19:
hi, since there is no information about how the lane markings are labeled in the argvoerse-v2 dataset. I wonder if these lane marking labels are annotated in the originally collected point cloud (labeling in 3D space), or if it is annotated on the image by projecting the point cloud onto the corresponding image.

hope you can help me figure out and thanks in advance :)

Mollylulu, 2022-09-27:
> Hi @Mollylulu ,
> 
> Thank you for your interest! Unfortunately, we cannot go into detail about the labeling process for the lane markings.
> 
> Ben

Noted, Ben. I don't want to know the details about the labeling processing, actually :) I just hope to have an idea about whether the annotation given is done on original point clouds (like point could segmentation labeling those points) or done on the 2d images (no details needed ).  I don't know if this information is accessible. if it is, hope you can tell me the short message. Thanks :)

James-Hays, 2022-09-27:
Sorry @Mollylulu , but even that level of detail we can't share.

Mollylulu, 2022-09-28:
I see, thanks :)

EchoQiHeng, 2023-07-12:
> I see, thanks :)

Are you interested in offline map annotations? ：）