argoverse/av2-api #98: "Control" information is missing in Argoverse 2 dataset

juyangbai, 2022-09-28:
Hi there, thank you for releasing this great dataset! 

I noticed the "has_traffic_control" information, which is in the Argoverse 1 datasets, is missing in the Argoverse 2 datasets. I'm building a motion prediction model that needs this information. So I'm wondering if you are planning to provide this information in the next releases. If so, when will the next version be released? 

FYI: The defination of "has_traffic_control" can be found in this [link](https://github.com/argoai/argoverse-api/issues/192).

kinnisoy, 2023-03-03:
Well, although it has been brought up, I would like to say that this issue also bothers me.

wqi, 2023-04-10:
Hey all, thanks for your interest in the AV2 datasets and apologies for the delayed response.

With the shutdown of Argo AI late last year, the Argoverse project is currently being maintained by a team of volunteers across industry and academia.

We acknowledge the inconvenience that not having `has_traffic_control` attributes might bring and will certainly include it in any future releases.

However, we don't currently have any firm estimates on a release date for an upgraded version of the AV2 maps.

closed 2023-04-10 by wqi