argoverse/av2-api #100: timestamps_ns in motion forecast dataset

sun1612, 2022-10-12:
I tried to convert timestamps_ns assuming epoch format and all scenarios seem to refer to date and time in the year 1980. Has there been any deliberate anonymization of the timestamp or am I doing the conversion wrong?

Thanks in advance!

Zhanwei-Z, 2023-03-13:
Hi, @sun1612 @benjaminrwilson @wqi I have the same question.  I wonder how to convert these timestamps_ns correctly!  
Thanks in advance!

betterqiang, 2023-03-15:
> Hi, @sun1612 @benjaminrwilson @wqi I have the same question. I wonder how to convert these timestamps_ns correctly! Thanks in advance!

hello， I just started to contact with data sets, but I don't know how to use AV2 at all. 

For example, every one in tutorial .py  how to use?       [data_ root_ path  ] [log_id]  How should I set in wind or linux?

This is my mailbox  imguoqiang@126.com. Can you communicate further if it is convenient

Thanks xiexie ya