argoverse/av2-api #102: Path issue in from_map_dir function of map_api

Shivanshu17, 2022-10-18:
The [vector_data_json_path variable ](https://github.com/argoai/av2-api/blob/05b7b661b7373adb5115cf13378d344d2ee43906/src/av2/map/map_api.py#L377) seems to extract the wrong path definition  (with relative path being passed in the [Map_Tutorial ](https://github.com/argoai/av2-api/blob/main/tutorials/map_tutorial.ipynb) notebook)

Setting it to just   `vector_data_fname`  seems to be working for me -- instead of    `log_map_dirpath  /  vector_data_fname`

Check it out please?

Thanks!

Abyssaledge, 2022-10-28:
I had the same problem and I solved it in the same way. The results look fine.
@Shivanshu17

ZLTJohn, 2023-01-08:
Thanks for reminder, @Shivanshu17 .  I think this is a bug since `log_map_dirpath.glob("log_map_archive_*.json")` will add dirpath to its result.