argoverse/av2-api #110: Map data in Motion Forecasting

Cram3r95, 2023-02-09:
Hi guys,

I am trying to use the map data in the Motion Forecasting dataset. Nevertheless, the file structure is a little bit different w.r.t. the Argoverse 1.1 Motion Forecasting dataset:

[image]

1. Should we assume the .json file is the local map around the focal agent in this particular sequence?
2. Where can I find similar functions to this one (find below) in Argoverse 1.1. which provides plausible centerlines?:

[image]

Cram3r95, 2023-02-15:
@benjaminrwilson @wqi

wqi, 2023-02-20:
Hey Carlos, apologies for the delayed response.

To answer your questions:

1. Yes, the `.json` file associated with each scenario contains the local vector map data in a bubble around the relevant actors.

2. We unfortunately don't have that specific function from AV1 ported over to the new API. If you'd like to see an (admittedly extremely rough) example of how this could be done for AV2, please see this [implementation](https://github.com/wqi/WIMP/blob/5bae0055c13ad03a27a1692f647c319cebdb49d0/scripts/map_features_utils.py#L152). 
That being said, it has been on my radar to get all of these functions ported cleanly over to the AV2 API eventually - will be top priority when I find the time.

Cram3r95, 2023-02-21:
@wqi thank you for your response. I also have these functions in my AV1 models:

https://github.com/Cram3r95/mapfe4mp/blob/498c1115df7e8fb3e42be9652975c5d262bf15fc/model/datasets/argoverse/map_functions.py#L839

But I mean for AV2, since the format is completely different. I think this can be really useful for the community.

wqi, 2023-02-21:
I know it's easy to miss at first glance, but the code I linked actually operates over AV2 maps and takes a `map_json` object. 

It's super rough because it was never intended for public release, but could serve as a reasonable stopgap in the meantime if you need reference code to work off of. We will get that functionality ported to the official API eventually, but we don't have a concrete ETA yet. 

With Argo winding down, AV2 has become an evening and weekends effort for most of us, so we appreciate the community's patience as we work to catch up with the backlog of issues and feature requests.

Cram3r95, 2023-02-22:
Hi @wqi, I almost finished to adapt your functions (regarding the Argoverse2 MAP Api) and your WIMP "argoverse-2.0" branch:

E.g. Considering this traffic scenario:

[image]

We can obtain the most plausible centerlines around a particular agent (in general the focal agent):

[image]

I think this will be helpful for the community.

closed 2023-03-30 by wqi