argoverse/av2-api #111: Is it possible to get the driving direction of a centerline?

AngelosSE, 2023-02-10:
Can LaneSegment.predecessors and LaneSegment.successors be used for this? Are these sorted according to the driving direction?

Cram3r95, 2023-02-15:
Hi @AngelosSE, I think we can estimate this with similar functions to Argoverse 1.1:

[    def get_candidate_centerlines_for_traj(](https://github.com/argoverse/argoverse-api/blob/f886ac54fba9f06f8a7d109eb663c7f501b3aa8e/argoverse/map_representation/map_api.py#L830)

AngelosSE, 2023-02-15:
Hi @Cram3r95, Thanks for the suggestion! I will look into it.

Cram3r95, 2023-02-15:
> Hi @Cram3r95, Thanks for the suggestion! I will look into it.

Actually I want to do pretty much the same, to obtain the centerlines most aligned with your trajectory, but there are no functions. If you want, we can collaborate.

AngelosSE, 2023-02-16:
I have currently put Argoverse2 on hold, and I am looking into NuScenes instead. I have found that NuScenes has a convenient API for accessing lane information, and I suggest that you have a look at it. When I get back to Argoverse2 and if I make any progress then I will let you know. :)

Cram3r95, 2023-02-16:
I think the map data is way better in Argoverse 2 than Nuscenes, the problem is that the map API is not developed yet at the same level than Argoverse 1, right @wqi?

wqi, 2023-02-22:
Hi @AngelosSE, apologies for the delayed response.

The predecessors and successors are indeed ordered based on driving direction. A lane segment's successor would be the next node along the direction of travel.

Re AV1 map API feature parity: there are a couple functions from the AV1 map API that weren't ported over to the AV2 API at release time, but we're hoping to add them in as there seems to be significant interest from the community.

If you have any other comments or suggestions regarding the AV2 scenarios and maps, please let us know! We're always looking to improve the user experience and understanding pain points helps us prioritize accordingly. :-)

closed 2023-02-22 by wqi