argoverse/av2-api #136: how get agent size in motion forecasting dataset?

betterqiang, 2023-03-24:
I can get agent P V ID in .parquet
but how to get it size?
Thanks

wqi, 2023-03-28:
Hi @betterqiang, ground truth information about actor dimensions (as inferred by tracking) is unfortunately not available in the current version of the AV2 MF dataset.

closed 2023-03-28 by wqi

James-Hays, 2023-03-29:
@betterqiang To add to what William said, even though the dataset doesn't have geometric dimensions for each actor, it does have the category of each actor, and there's not that much variance in size within each category. E.g. if you use the average dimensions of a pedestrian, a car, a cyclist, etc I think it will be a pretty good approximation.

betterqiang, 2023-04-27:
Thank you for the answer.
 I obtained the  size of the agent from the Sensor dataset and provided an approximate size.