argoverse/av2-api #150: Submission and local evaluation of End-To-End Forecasting

foolhard, 2023-04-06:
Hi, 

I cannot find the tutorials for submission and local evaluaiton of End-to-end forecasting task. When will you release any turorials or sample codes of these funtions?

Besides, this task has 50 meter range detection. If some objects are partially within 50m, will it be counted in or removed? 

Thanks.

neeharperi, 2023-04-11:
Hi @foolhard, 
Thanks for your interest in the end-to-end forecasting task. Please check out our baselines at https://github.com/neeharperi/LT3D and let me know if you have any further questions. We'll continue to add additional baselines in the coming weeks.

Neehar

foolhard, 2023-04-13:
@neeharperi Thanks for your feedback. I will try your baseline code.