argoverse/av2-api #153: CVPR2023 motion challenge, can not submit the result parquet file

kingwmk, 2023-04-10:
Hello. Thanks for your great contributions!!!
I can not submit the result parquet file (305 M) successfully on evalai.
When I using CLI, the uploading process always terminated with following error. Even though I switched many networks.

19%|███████████████▌                                                                   | 3/16 [04:51<21:02, 97.15s/it]

There was an error while uploading the file: HTTPSConnectionPool(host='evalai.s3.amazonaws.com', port=443): Max retries exceeded with url: /media/submission_files/submission_259211/820b0aa5-3194-4b44-92f2-fc3d64745531.parquet?uploadId=w46EJjKLs.PoGxdOs7TTDYBgXrKVk4MT7Jeqqtk9lvKxxtW.1dhu8z72SdEa8FGdtkBgdSZYXQ2GUAZvuCGTPMA8G67.7uVbOa5sEhh825QUQk64QAfjHanPJ04rd1UD&partNumber=4&AWSAccessKeyId=AKIAIVU6WOHPESDHAVZA&Signature=NynjwkCqKzTuVvAgRYqWgwCe%2FSU%3D&Expires=1681144663 (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:2396)')))

wqi, 2023-04-12:
Hi @kingwmk, this appears to be an issue with the EvalAI platform.

I do see your submission being created on the backend, but it looks like the file contents are empty due to that upload error.

Could you please try submitting through the web interface? Let me know if that works. :-)

ZikangZhou, 2023-04-17:
> Hi @kingwmk, this appears to be an issue with the EvalAI platform.
> 
> I do see your submission being created on the backend, but it looks like the file contents are empty due to that upload error.
> 
> Could you please try submitting through the web interface? Let me know if that works. :-)

I tried to submit a baseline to EvalAI, but it seems that uploading files via the web interface always gets stuck. Could you help confirm whether uploading files works well on your side?

kingwmk, 2023-04-17:
>but it seems that uploading files via the web interface always gets stuck. 

I got stuck here too. And the results submitted through the evalai CLI have blocked my account. These result files have been submitted, but they have not been successfully evaluated or failed. So, my account can not submit new result files through the evalai CLI.

wqi, 2023-04-17:
Hi folks, I've cleared out the EvalAI submission queue for the challenge. If you check your account again, you should be able to generate new submissions now.

Submitting via web interface does work for me, with metrics being returned successfully.

We're continuing to work with the EvalAI team to root cause the problem, but their current working theory is that this is an issue with the boto3 API used as part of CLI upload. When uploading large files in a chunked manner, S3 sometimes drops connections leading to upload failure. I'm being told that this can happen when users have unreliable internet, but unfortunately don't have any further information at this time.

@ZikangZhou @kingwmk I'll reach out individually tomorrow to arrange a temporary workaround. Sorry about all the difficulties you've been facing so far, I know it's not an ideal experience. :-(

ZikangZhou, 2023-04-17:
@wqi,

Thanks for your help :) Now I can upload files via CLI. I can confirm that the test set results are pretty close to my local evaluation results on the validation set. But one thing to note is that you may have unintentionally switched avgMinFDE (K=1) and avgMinFDE (K=6). The avgMinADE has the same issue. Hope this information useful to you.

kingwmk, 2023-04-17:
> folks

Thanks!!! it works!!

wqi, 2023-04-18:
Thanks for catching that @ZikangZhou! Evaluation backend has been updated, those metrics should be reported with the correct labels now.

kangmiao618, 2023-05-30:
Hi, @wqi , I cann't submit the result through the web interface too. Is there any other way to submit the result?

wqi, 2023-05-30:
Hi @kangmiao618, I can confirm this is an issue with the evaluation server.

Working to resolve the issue now, will post an update once things are back online.

Thank you for your patience.

kangmiao618, 2023-05-31:
@wqi Thanks for your reply. Now I could upload the file with CLI  and I can see my submission on the web. But I didn't get any result file. Uploading the files from the web interface is still stucked now.

wqi, 2023-05-31:
Just pushed through all the stuck submissions - everything should be good now!

Please feel free to re-open this issue if you run into problems again.

closed 2023-05-31 by wqi

kangmiao618, 2023-05-31:
@wqi Thanks! It worked.

yenw, 2023-07-08:
Hi @wqi, Thanks for your help. 
The status is always "Submitted" in "Single Agent Forecasting".

[image]

xjh199923, 2023-11-29:
@wqi Hi, I cann't submit the result through the web interface and use cli too.  Uploading in cli shows the following error "Error: {'detail': 'Given token not valid for any token type', 'code': 'token_not_valid', 'messages': {'token_class': 'RefreshToken' , 'token_type': 'refresh', 'message': 'token is invalid or expired'}]}

Use `evalai challenges` to get active challenges.

Use `evalai challenge CHALLENGE phases` to get the active phases.
"

[image]

wqi, 2023-12-05:
Hi @xjh199923, sorry about the trouble. EvalAI was experiencing some connectivity issues last week which was likely the root cause of the issues you were seeing.

If you're still seeing issues on your end, please let us know and we'll continue investigating.

XinGP, 2024-12-13:
> @wqi Hi, I cann't submit the result through the web interface and use cli too. Uploading in cli shows the following error "Error: {'detail': 'Given token not valid for any token type', 'code': 'token_not_valid', 'messages': {'token_class': 'RefreshToken' , 'token_type': 'refresh', 'message': 'token is invalid or expired'}]}
> 
> Use `evalai challenges` to get active challenges.
> 
> Use `evalai challenge CHALLENGE phases` to get the active phases. "
> 
> <img alt="image" width="1841" src="https://private-user-images.githubusercontent.com/45587237/286499856-8118eea4-0f8c-4b46-a6d0-b8c85605f704.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQwNzc2ODAsIm5iZiI6MTczNDA3NzM4MCwicGF0aCI6Ii80NTU4NzIzNy8yODY0OTk4NTYtODExOGVlYTQtMGY4Yy00YjQ2LWE2ZDAtYjhjODU2MDVmNzA0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDA4MDk0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0OTA4YzY1NDdlNTRmNDYwNjRhZjA5Y2U3MDE2MGJmOWEwNTQ4OGE3ZDlkNzdjYTY3M2JmNTA3YzEzZWFjNmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yhoaV5Zxrz8zAVNEyEZO_P83qqrPHVOAWCkCbZjwSLU">

@wqi I encountered the same problem when I used the CLI, and it was also stuck on multiple Web machines. Can you check again to see if the server has a problem?

wqi, 2024-12-13:
This looks like an EvalAI authentication issue... you may have to check that you have a valid API token. If you're still having trouble after checking the documentation and FAQs, it may be worth reaching out the EvalAI support team.

I've verified AV2 evaluation server is currently online and processing submissions without issue.