argoverse/av2-api #154: Multi-Agent Motion Forecasting: Unclear Description of Metrics and Evaluation Protocol

ZikangZhou, 2023-04-12:
Hi @wqi,

Thanks for the efforts on the multi-agent motion forecasting benchmark!

I have some doubts about the metrics and the evaluation protocol. First, it is said that "predictions will be scored for all 'scored actors' marked in each scenario." However, the meaning of "scored actors" may be confusing. Does it mean the tracks with the category of "SCORED_TRACK"?  Or does it mean the union of "SCORED_TRACK" and "FOCAL_TRACK"?

The multi-world metrics are also ambiguous (mainly avgMinADE and avgBrierMinFDE). The unclear part is about the criterion of selecting the world-level minimum value. For avgMinFDE, actorMR, and actorCR, the "best world" is defined as the one with the lowest meanFDE. But for avgMinADE and avgBrierMinFDE, it is unclear whether the "best world" has the same definition as other metrics.

wqi, 2023-04-12:
Hi @ZikangZhou, thanks for reaching out!

We always appreciate it when AV2 users post their questions on GitHub, so we can figure out where friction might exist within our datasets and challenges.

To answer your questions:

1. The definition of "scored actors" includes all tracks labeled with `SCORED_TRACK` and `FOCAL_TRACK` - it's a superset of the tracks evaluated in the standard single agent challenge.
2. Similar to the single-agent challenge, the predicted world with the lowest avgADE is reported as the avgMinADE. To compute the summary metric, we take the mean avgMinADE across all evaluted scenarios in the test set. 
3. To report the avgBrierMinFDE for each scenario, we adopt a similar approach to actorMR and actorCR, by taking the avgBrierFDE from the world with the lowest avgFDE.

I'll update the challenge page on EvalAI to clear things up, but hope this helps in the meantime! :-)

ZikangZhou, 2023-04-12:
Thanks for the clarification!

closed 2023-04-12 by ZikangZhou

kangmiao618, 2023-05-31:
Hi @wqi , Sorry to bother you again. I occured a problem when I write the submission parquet of  multi-agent forecasting. I'm cofused with the define of PredictionProbabilities in src/av2/datasets/motion_forecasting/eval/submission.py #Line 19. My model produce 6 trajectories (6, 60, 2) and 6 scores(6) for each scored actors with a unique track_id in a scenario . How could I write them to the submission file. Because the defined input of the ChallengeSubmission is required to Dict[Scenario_ID, ScenarioPredictions ] -> ScenarioPredictions = Dict[track_id, TrackPredictions]->TrackPredictions = Tuple[PredictedTrajectories, PredictionProbabilities]. This means I can only add one scored actor for one scenario.

wqi, 2023-05-31:
Hi @kangmiao618, a set of `ScenarioPredictions` is defined [here](https://github.com/argoverse/av2-api/blob/main/src/av2/datasets/motion_forecasting/eval/submission.py#L21) to be a tuple of `ScenarioProbabilities` and `ScenarioTrajectories`.

The submission format was updated with the latest API version to accommodate predictions for multiple actors in each scenario. Updating your API to the latest version on GitHub should resolve your issues with generating the submission file. :-)