argoverse/av2-api #167: i want to visualization the dataset like Argoverse1

Syedaliraza005, 2023-04-24:
i want to visualization the dataset like Argoverse1. I have downloaded the dataset and want to visualize the sensor dataset 
like Argoverse1 .I did not find any tutorial for that please help me out  
[image]

Syedaliraza005, 2023-04-24:
step by step help needed

argoverse-admin, 2023-12-17:
Hi @Syedaliraza005, thanks for your interest in Argoverse 2. We provide one script here to visualize annotated cuboids on images: https://github.com/argoverse/av2-api/blob/main/tutorials/generate_sensor_dataset_visualizations.py

It will generate visualizations that resemble the gifs shown here: https://argoverse.github.io/user-guide/datasets/sensor.html