argoverse/av2-api #181: Multi-Agent Motion Forecasting:  Data inconsistency occurred in EvalAI and dataset downloaded from website

ares89, 2023-05-06:
hi @wqi :
When I submitted my results to EvalAI, it returned an error message: 
KeyError('No predictions found for track 85907 in scenario 0e4e3274-5a8f-41cc-b3da-985a87ed8108.'). 
However, upon checking the data, it shows that the track category of track 85907 in scenario 0e4e3274-5a8f-41cc-b3da-985a87ed8108 is UNSCORED_TRACK. 
How can I resolve this issue? Thank you!
[image]

ares89, 2023-05-06:
Does this dataset have different versions?

wqi, 2023-05-08:
Hi @ares89, the dataset was updated once last year.

Which track are you trying to submit to and how recently did you download the dataset?

ares89, 2023-05-09:
Thank you for your reply. 
The track I am trying to submit is multi-word forecasting.
The dataset was downloaded approximately ten months ago, in August of 2022.

wqi, 2023-05-09:
I'd recommend downloading the dataset again from the website - should take care of the issue. :-)