argoverse/av2-api #196: What does the term "multi-world" mean?

ares89, 2023-05-22:
@wqi hello, what does the term "multi-world" mean? I have been unable to find a specific definition or reference to it.
I guess that, one scenario only have six wolrds, every world has one probability.

ares89, 2023-05-22:
I found the new submission format of multi-world motion-forecasting, is this using in evalai now?

wqi, 2023-05-30:
Hi @ares89, the term "multi-world" is used in the context of AV2 to differentiate this year's motion forecasting challenge setup (which requires predictions for multiple actors in each scenario) from last year's - which requires predictions to be submitted only for the focal agent (hence "single-agent" prediction).

Each "world" refers to a single predicted joint future capturing the state of all actors in the scene. This differs from an alternative multi-agent prediction setup where the predicted trajectories for each actor are scored individually.

closed 2023-05-30 by wqi