argoverse/av2-api #199: argoverse2 tracking benchmark upload error

lejk8104, 2023-05-30:
Dear Argo,

I am reaching out to inform you about an issue I encountered while using your valuable service for the Argoverse 2: End-to-End Forecasting Challenge, in which I attempted to evaluate the performance of my tracking model (Evaluate about Tracking section).

[image]

In Stderr file log
`Traceback (most recent call last):
  File "remote_challenge_evaluation/run_worker.py", line 86, in score_submission
    eval_out = evaluate(test_annotation_path,
  File "/home/nperi/Workspace/AV2-Challenge/av2-e2e-prediction-challenge-evalai/evaluation_script/main.py", line 75, in evaluate
    _, tuned_hota_per_class, _ = evaluate_tracks(prediction, ground_truth, "HOTA", max_range_m, dataset_dir, "")
  File "/home/nperi/Workspace/Core/av2-api/src/av2/evaluation/tracking/eval.py", line 525, in evaluate
    utils.annotate_frame_metadata(
  File "/home/nperi/Workspace/Core/av2-api/src/av2/evaluation/tracking/utils.py", line 55, in annotate_frame_metadata
    assert len(prediction_frames) == len(label_frames)
AssertionError`

I came across identical error messages in both the test set and the validation set. I used the "sensor dataset" downloaded for the evaluation, and the count of frames in the pkl file I uploaded - the outcome of len(prediction_frames) - was 23574 and 23547 for the validation and test sets respectively.

In light of this, I am curious if you could provide precise information about the total quantity of the tracking validation and test data.

I look forward to your response and appreciate your help in advance.

neeharperi, 2023-05-30:
Hi @lejk8104,
The validation set is of length 2404. I have added a print statement with this information to the assert statement to make it easier to debug.
Please see our [documentation](https://argoverse.github.io/user-guide/tasks/e2e_forecasting.html#tracking)  for more information on the expected format, or this [sample tracking result](https://drive.google.com/file/d/122tPioHzyBiZU349pqy8KDg9b2wUGC49/view?usp=sharing) for the validation set. Please let me know if you have any further questions. 

Neehar

lejk8104, 2023-05-31:
Thank you for your kind response. I will try to resolve the issue by checking the sample.

Could you possibly inform me about the deadline for the CVPR2023 workshop Tracking Challenge?

neeharperi, 2023-05-31:
All AV2 CVPR competitions will close on June 11th 11:59 PM EST

lejk8104, 2023-06-01:
I would like to express my gratitude for the sample data you provided, which enabled me to effectively resolve the previous issue I was facing.

Q1. However, I must inform you that a new error has occurred during the server upload process.
# our error msg
[image]

I attempted to debug this error, but found that the relevant part of the code has been changed.
# latest api
[image]

Q2. In light of this, could you kindly confirm if the tracking eval code provided in the challenge I am currently participating in differs from the eval code shared on the argo2-api GitHub?

The atta
ched image contains information about the sample data provided by you and my data.

# your samples
[image]
# our sample
[image]
s
[image]

As always, I am grateful for your assistance.

neeharperi, 2023-06-01:
Q1: It seems that the same issue you had previously is still triggering the same assert statement on L55. Unfortunately I can't see any obvious issues with the screenshots you shared. Can you confirm that both your submission and the sample tracking file I provided evaluate correctly on your local machine? 
Q2: I can confirm that the evaluation code running on our server is identical to the av2-api

neeharperi, 2023-06-02:
I think I've figured out the issue. About two months ago, we pushed a small fix (See [Issue](https://github.com/neeharperi/LT3D/issues/4#issuecomment-1523299729)) that corrected the sampling rate. Both your tracking results and my tracking results that I shared with you were generated using an older version of the av2_infos_trainval.pkl file.  If you pull the latest version of LT3D, or make the small fix described in the above issue, I think you should be able to successfully submit to EvalAI. 

I'm regenerating the new av2_infos_trainval.pkl file on my local machine to validate this as well. Please let me know if this solves your issue.

lejk8104, 2023-06-03:
By referring to the link you provided and adjusting the sampling cycle from 10 to 5, the test on Eval AI was successfully conducted. Your detailed and kind response was of great assistance in resolving this issue. Thank you very much.

closed 2023-06-03 by lejk8104