argoverse/av2-api #204: pointcloud motion compensation

Kailthen, 2023-06-12:
1 is the [x,y,z] fields in pointcloud undistorted ?
2 if not, how can i do this myself ?

thanks

Kailthen, 2023-06-13:
From https://argoverse.github.io/user-guide/datasets/sensor.html,
Both lidars have their own reference frame, and we refer to them as up_lidar and down_lidar, respectively. We have **egomotion-compensated** the LiDAR sensor data to the egovehicle reference nanosecond timestamp

closed 2023-06-13 by Kailthen

Wolfybox, 2024-08-08:
> From https://argoverse.github.io/user-guide/datasets/sensor.html, Both lidars have their own reference frame, and we refer to them as up_lidar and down_lidar, respectively. We have **egomotion-compensated** the LiDAR sensor data to the egovehicle reference nanosecond timestamp

I notice that if the timestamp is larger, mere egomotion-compensated is not enough to align the bbox. Maybe taking the obj motion into consideration as well would solve this?