argoverse/av2-api #205: Global coordinate frame to GPS

tobiasfshr, 2023-06-14:
Is it possible to convert the global, city-scale poses to GPS coordinates? This would be helpful to visualize the covered area and potentially include other data sources such as OpenStreetMap.

EchoQiHeng, 2023-07-11:
Friend, have you achieved it yet？

tobiasfshr, 2023-09-08:
awesome thanks!

closed 2023-09-08 by tobiasfshr

GoroYeh-HRI, 2024-04-30:
> Hi, we provide instructions on how to do this here: #108?
> 
> Please see the per-city origins and utility functions defined here: https://github.com/argoverse/av2-api/blob/main/src/av2/geometry/utm.py#L53
> 
> ```python
> def convert_gps_to_utm(
>     latitude: float, longitude: float, city_name: CityName
> ) -> Tuple[float, float]:
>     """Convert GPS coordinates to UTM coordinates.
> ```
> 
> ```python
> def convert_city_coords_to_wgs84(
>     points_city: Union[NDArrayFloat, NDArrayInt], city_name: CityName
> ) -> NDArrayFloat:
>     """Convert city coordinates to WGS84 coordinates.
> ```
> 
> Map data and poses are provided in each city's coordinate frame.
Thank you so much for this information! I appreciate it.
May I ask what are the differences between `UTM` and `WGS84` coordinates?
If I want to convert an (x, y) point in a HD map to (lat ,lng) on the globe, is it possible?
Or vice versa, converting a (lat, lng) to (x, y) in a HD map in Argoverse2.
Thank you!

GoroYeh-HRI, 2024-04-30:
Also, let's say I have a (N,2) array representing in (lat, lng) WGS84 coordinates,
how do I convert that to a specific "city coordinates"?
For example, if I want to convert it to city 'Palo Alto'. How do I convert the coordinates from (lat, lng) to (x, y) with respect to Palo Alto's coordinate frame?

argoverse-admin, 2024-05-01:
Hi @GoroYeh-HRI, thanks for your interest in Argoverse 2.0.

Please refer to Appendix A.1 of our [Argoverse V1 paper](https://arxiv.org/pdf/1911.02620):
```
A.1. Coordinate System
The model of the world that we subscribe to within our map
and dataset is a local tangent plane centered at a central point located 
within each city. This model has a flat earth assumption
which is approximately correct at the scale of a city. Thus, we
provide map object pose values in city coordinates. City coordinates 
can be converted to the UTM (Universal Transverse Mercator) 
coordinate system by simply adding the city’s origin in UTM
coordinates to the object’s city coordinate pose. The UTM model
divides the earth into 60 flattened, narrow zones, each of width
6 degrees of longitude. Each zone is segmented into 20 latitude
bands.
```
Since we provide the origin of each city coordinate system in the UTM cartesian coordinate frame:

You can do convert the GPS to UTM, and then subtract the UTM coords of the city origin, to get city coordinates. Reverse engineering the ops here https://github.com/argoverse/av2-api/blob/main/src/av2/geometry/utm.py#L43, you would get something like

```
def convert_wgs84_points_to_city_coords(
    points_wgs84: Union[NDArrayFloat, NDArrayInt], city_name: CityName
) -> NDArrayFloat:
    """Convert WGS84 coordinates to city coordinates.

    Args:
        points_wgs84: Array of shape (N,2), representing points in the WGS84 coordinate system, as (latitude, longitude).
        city_name: Name of city, where query points are located.

    Returns:
        2d points in city coordinates, as (N,2) array.
    """
    latitude, longitude = CITY_ORIGIN_LATLONG_DICT[city_name]
    # Get (easting, northing) of origin.
    origin_utm = convert_gps_to_utm(
        latitude=latitude, longitude=longitude, city_name=city_name
    )
     
    points_city = np.zeros_like(points_wgs84)
    for i, (lat, long) in enumerate(points_wgs84):
      point_utm = convert_gps_to_utm(
          latitude=lat, longitude=long, city_name=city_name
      )
      points_city[i] = point_utm - origin_utm

    return points_city
```