argoverse/av2-api #211: What is the reason for the error that occurred during the installation of AV2 API?

EchoQiHeng, 2023-07-07:
Building wheels for collected packages: av2
  Building editable for av2 (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building editable for av2 (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [55 lines of output]
      Running `maturin pep517 build-wheel -i /home/qh/anaconda3/envs/av2/bin/python --compatibility off --editable`
      📦 Including license file "/home/qh/av2-api/NOTICE"
      🍹 Building a mixed python/rust project
      🔗 Found pyo3 bindings
      🐍 Found CPython 3.8 at /home/qh/anaconda3/envs/av2/bin/python
      📡 Using build options features from pyproject.toml
         Compiling autocfg v1.1.0
         Compiling libc v0.2.141
         Compiling proc-macro2 v1.0.56
         Compiling quote v1.0.26
         Compiling unicode-ident v1.0.8
         Compiling cfg-if v1.0.0
         Compiling libm v0.2.6
         Compiling scopeguard v1.1.0
         Compiling syn v1.0.109
         Compiling crossbeam-utils v0.8.15
         Compiling memchr v2.5.0
         Compiling futures-core v0.3.28
         Compiling once_cell v1.17.1
         Compiling serde_derive v1.0.160
         Compiling version_check v0.9.4
         Compiling futures-sink v0.3.28
         Compiling adler v1.0.2
         Compiling serde v1.0.160
         Compiling static_assertions v1.1.0
         Compiling log v0.4.17
         Compiling rayon-core v1.11.0
         Compiling either v1.8.1
         Compiling crc32fast v1.3.2
         Compiling miniz_oxide v0.6.2
         Compiling pkg-config v0.3.26
         Compiling futures-channel v0.3.28
         Compiling num-traits v0.2.15
         Compiling memoffset v0.8.0
         Compiling crossbeam-epoch v0.9.14
         Compiling num-integer v0.1.45
         Compiling lock_api v0.4.9
      error: could not compile `proc-macro2` (lib) due to 2 previous errors
      warning: build failed, waiting for other jobs to finish...
      💥 maturin failed
        Caused by: Failed to build a native library through cargo
        Caused by: Cargo build finished with "exit status: 101": `PYO3_ENVIRONMENT_SIGNATURE="cpython-3.8-64bit" PYO3_PYTHON="/home/qh/anaconda3/envs/av2/bin/python" PYTHON_SYS_EXECUTABLE="/home/qh/anaconda3/envs/av2/bin/python" "cargo" "rustc" "--release" "--features" "pyo3/extension-module" "--manifest-path" "/home/qh/av2-api/rust/Cargo.toml" "--message-format" "json" "--lib" "--crate-type" "cdylib"`
      error[E0635]: unknown feature `proc_macro_span_shrink`
        --> /home/qh/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.56/src/lib.rs:92:30
         |
      92 |     feature(proc_macro_span, proc_macro_span_shrink)
         |                              ^^^^^^^^^^^^^^^^^^^^^^
      
      
      error: aborting due to previous error
      
      
      For more information about this error, try `rustc --explain E0635`.
      
      Error: command ['maturin', 'pep517', 'build-wheel', '-i', '/home/qh/anaconda3/envs/av2/bin/python', '--compatibility', 'off', '--editable'] returned non-zero exit status 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for av2
Failed to build av2
ERROR: Could not build wheels for av2, which is required to install pyproject.toml-based projects

EchoQiHeng, 2023-07-08:
Resolved. "cargo update"

Qingfeng800, 2023-07-11:
@EchoQiHeng Hi, I meet the same problem with you, when I try to run "cargo update" in conda enviroment, raise another problem:error: could not find `Cargo.toml` in `/home/qf/WORK/QCNet` or any parent directory. Do you know how to solve it, my Rust and Cargo is installed in "/home/qf/.cargo" .

EchoQiHeng, 2023-07-11:
> @EchoQiHeng Hi, I meet the same problem with you, when I try to run "cargo update" in conda enviroment, raise another problem:error: could not find `Cargo.toml` in `/home/qf/WORK/QCNet` or any parent directory. Do you know how to solve it, my Rust and Cargo is installed in "/home/qf/.cargo" .

git clone https://github.com/argoverse/av2-api
cd av2-api
cargo update

Qingfeng800, 2023-07-11:
@EchoQiHeng Thanks! I have "cargo update" successfully. But after "cargo update", I run "pip install git+https://github.com/argoverse/av2-api#egg=av2" to install and still got "ERROR: Failed building editable for av2" .How did you solve it at last?

EchoQiHeng, 2023-07-11:
> 

git clone https://github.com/argoverse/av2-api
pip install -e /path_to_root_directory_of_the_repo/

closed 2023-07-12 by EchoQiHeng