argoverse/av2-api #212: error[E0635]: unknown feature `proc_macro_span_shrink` while building av2 by Pip

eyecod-toppick, 2023-07-10:
My python version is 3.8.17.
I installed av1 before. When I try to install av2 using pip, I get an error when compiling av2：

```
      error: could not compile `proc-macro2` (lib) due to 2 previous errors
      warning: build failed, waiting for other jobs to finish...
      💥 maturin failed
        Caused by: Failed to build a native library through cargo
        Caused by: Cargo build finished with "exit status: 101": `PYO3_ENVIRONMENT_SIGNATURE="cpython-3.8-64bit" PYO3_PYTHON="/usr/local/anaconda3/envs/nasTNT/bin/python" PYTHON_SYS_EXECUTABLE="/usr/local/anaconda3/envs/nasTNT/bin/python" "cargo" "rustc" "--release" "--features" "pyo3/extension-module" "--manifest-path" "/home/yipin/program/nas/nasTNT/av2-api/rust/Cargo.toml" "--message-format" "json" "--lib" "--crate-type" "cdylib"`
      error[E0635]: unknown feature `proc_macro_span_shrink`
        --> /home/yipin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.56/src/lib.rs:92:30
         |
      92 |     feature(proc_macro_span, proc_macro_span_shrink)
         |                              ^^^^^^^^^^^^^^^^^^^^^^
      
      
      error: aborting due to previous error
      
      
      For more information about this error, try `rustc --explain E0635`.
      
      Error: command ['maturin', 'pep517', 'build-wheel', '-i', '/usr/local/anaconda3/envs/nasTNT/bin/python', '--compatibility', 'off', '--editable'] returned non-zero exit status 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for av2
Failed to build av2
ERROR: Could not build wheels for av2, which is required to install pyproject.toml-based projects
```

eyecod-toppick, 2023-07-10:
I update my Cargo and the problem fixed

Qingfeng800, 2023-07-11:
@eyecod-toppick Hi, I meet the same problem with you, my Cargo is the latest version, but still have the problem, which version of Cargo is yours?

Qingfeng800, 2023-07-11:
@eyecod-toppick Hi, how do you update Cargo, is it via "rustup update stable"?

Qingfeng800, 2023-07-11:
@eyecod-toppick Hi,after "cargo update", how did you install av2 later? Is it installed according to "pip install git+https://github.com/argoverse/av2-api#egg=av2" ? I still meet the same problem after "cargo update".

ziqipang, 2023-07-12:
@eyecod-toppick Could you let me know your current version? I am having the same issue. Thank you so much!

ziqipang, 2023-07-12:
Problem solved. Sorry for the interrupt

USCT-YQJ, 2023-07-17:
> Problem solved. Sorry for the interrupt

how did you solve the problem? thank you~

shengchao-y, 2023-07-27:
> > Problem solved. Sorry for the interrupt
> 
> how did you solve the problem? thank you~

you can refer to this answer.
https://github.com/argoverse/av2-api/issues/211#issuecomment-1626658908