argoverse/av2-api #214: Annotations for High-precision Maps

EchoQiHeng, 2023-07-21:
Hello, I am currently working on high-precision map generation based on the AV2 dataset, and my confusion lies in the annotation of high-precision maps. As shown in the following figure, map elements within the sensor range (point cloud map area) can be easily labeled based on the movement trajectory of the vehicle, while the problem is **how to label other areas (not observed by LiDAR)**. I want to do some high-precision map automatic generation tasks. Do I need to manually select point and line elements within the range of the point cloud map?
[image]
[image]

James-Hays, 2023-07-21:
Hi!

The HD map elements paired with each scenario are intentionally sampled from a wide radius. Some of those HD map elements are probably not observable from lidar or from camera. So it isn't a realistic task to try and generate the entire HD map for a scenario given only its sensor data. Argo generated the maps by looking at lots of data that isn't part of this scenario. 

If you're trying to establish a benchmark for map generation, you'd probably want some rule such as "only train or evaluate on map elements within 25 meters of the ego vehicle" or maybe even include some explicit notion of visibility e.g. "only train or evaluate on map elements within 50 meters of the ego vehicle and that have a least 1 lidar return within 1 meter" or something like that.

We haven't established an HD map synthesis benchmark. We did, however, establish an HD map verification benchmark with the Trust, but Verify dataset: https://www.argoverse.org/av2.html#mapchange-link

EchoQiHeng, 2023-07-23:
> Hi!
>
> The HD map elements paired with each scenario are intentionally sampled from a wide radius. Some of those HD map elements are probably not observable from lidar or from camera. So it isn't a [...]

  Thank you very much for your reply. As you mentioned, I am committed to building high definition maps (online/offline). I will follow your suggestions and manually set some rules to establish map benchmarks.

   I am also interested in the TbV dataset, but when I visualized the map, I noticed some annotation issues. As shown in the following figure, for an intersection in the scene, the purple area represents a pedestrian crossing, but the pedestrian crossing of the left road is not included in the GT. Therefore, I visualized the point cloud (reflection intensity) and RGB point cloud, and it is clear that the left road contains a pedestrian crossing. 

  May I ask if this is an omission in the labeling or my understanding is incorrect? Looking forward to your reply!

/home/qh/图片/7.23/Figure_1.png
/home/qh/图片/7.23/intensity.png
/home/qh/图片/7.23/rgb.png

EchoQiHeng, 2023-07-23:
> Hi!
>
> The HD map elements paired with each scenario are intentionally sampled from a wide radius. Some of those HD map elements are probably not observable from lidar or from camera. So it isn't a [...]

Thank you very much for your reply. As you mentioned, I am committed to building high definition maps (online/offline). I will follow your suggestions and manually set some rules to establish map benchmarks.

I am also interested in the TbV dataset, but when I visualized the map, I noticed some annotation issues. As shown in the following figure, for an intersection in the scene, the purple area represents a pedestrian crossing, but the pedestrian crossing of the left road is not included in the GT. Therefore, I visualized the point cloud (reflection intensity) and RGB point cloud, , and it is clear that the left road includes a pedestrian crossing.

May I ask if this is an omission in the labeling or my understanding is incorrect? Looking forward to your reply!

[image]
[image]
[image]

DaliMIT01, 2024-12-19:
@EchoQiHeng hello, how to generate this kind of image?
[image]

DaliMIT01, 2024-12-19:
@EchoQiHeng hello,how to generate this kind of image? Which script was used?
[image]

EchoQiHeng, 2024-12-22:
> @EchoQiHeng hello,how to generate this kind of image? Which script was used? ![下载](https://private-user-images.githubusercontent.com/150243153/397396973-a51aaf00-d0cb-45a9-a12f-8e235d75a289.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4NTE3OTgsIm5iZiI6MTczNDg1MTQ5OCwicGF0aCI6Ii8xNTAyNDMxNTMvMzk3Mzk2OTczLWE1MWFhZjAwLWQwY2ItNDVhOS1hMTJmLThlMjM1ZDc1YTI4OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMlQwNzExMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yOWVmNzAzMTBlNGUxODgxMjA1OWE4MjRkYzFmZTlhMmIxZTgxZmY2ZTQ2MjllMDFjZjY2MGIyNzY5MzIzNjEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.a2YWOuqCrVqrjZImq0MdjQMI_Mdcrw-XuKbutWpbwHs)

The official tutorial for NUSCENES.