argoverse/av2-api #215: Calibration Method and Recording?

aryasenna, 2023-07-26:
Hello!

Can anyone tell us how did Agroverse2 sensor got calibrated? especially for intrinsic camera and extrinsic Lidar-camera
The paper does not cover this topic. 

I'm interested what kind of accuracy (e.g., reproj error) you achieved for the Perception Dataset, and checkerboard recording.

aryasenna, 2023-07-27:
hi @benjaminrwilson 

Thanks for coming back, I see there is no reproj error at the moment, is it possible to publish the checkerboard recording so the public can calibrate?

For my (academic) research, I'm checking if my calib method is better than whatever is used for Agroverse and then publish 
the method.

James-Hays, 2023-07-27:
Hi @aryasenna 

We don't have the original calibration sequences to release.

Calibration is important and difficult to get right. We think the Argoverse 2 calibration is pretty good, but I don't doubt there is room for improvement on certain scenarios. We're open to someone suggesting an improved calibration. It would need to be based on analysis of the particular driving scenarios, not a calibration pattern.