argoverse/av2-api #223: Clarification and Inquiry about Argoverse Motion Forecasting Dataset

jaeaehun, 2023-08-30:
Is the track_id "AV" in the parquet file https://www.argoverse.org/av2.html#forecasting-link the green ego vehicle described here?

There was a phrase "possible forecasting targets are indicated in orange". Then is there a separate track_id that can be learned (predictable) in the parquet puller? If there is a separate track_id that can be predicted (learned), what are the criteria?

There are two states in the parquet file whether observed is true or false. Is this true or false if the ego vehicle is within the sensing range? It seems that a specific explanation is needed. The reason for the confusion is that even if observed is false, there is position and yaw information.

I am drawing a new map with a json file to apply my algorithm. Do I need to grab my view based on focal_id to draw like the link above here? Or should I get a view based on ego_vehicle?
Are possible forecasting targets relevant?

jaeaehun, 2023-09-05:
and i don't understand "observed" data-  True and False

wqi, 2024-02-27:
Hi @jaeaehun, the track ID `AV` does correspond to the green vehicle in the linked figure.

Each track belongs to one of [four categories](https://github.com/argoverse/av2-api/blob/main/src/av2/datasets/motion_forecasting/data_schema.py#L18): track fragment, unscored track, scored track, or focal track. If a track is marked as either a scored track or the focal track for a scenario, predictions for that actor will be scored by the evaluation system.

The term "observed" refers to whether a particular object state falls in the observed segment of a scenario (as opposed to the predicted segment of a scenario). You can think of objects states with `observed=True` as being part of the inputs for a particular scenario, while `observed=False` would be part of the ground truth labels.

closed 2024-02-27 by wqi