argoverse/av2-api #227: Meaning of lidar `offset_ns`

BenAgro314, 2023-09-29:
In the documentation for the provided LiDAR data, there is a field `offset_ns`

[image]

If I wanted to get the exact timestamp each lidar point, would I add `offset_ns` to the reference timestamp, or would I subtract it?

BenAgro314, 2023-09-29:
Coloring the lidar points from `up_lidar` based on their timestamp in a single sweep: 
[image]
where dark blue is 0 and yellow is 0.1 s.

So this would indicate that, looking from the top down, the VLP-32 LiDAR was spinning clockwise. 
Is that the case?

James-Hays, 2023-10-18:
I think you've got it right, yes.