argoverse/av2-api #229: Motion prediction submitssion

soyons, 2023-10-14:
I am newer to the challenge and  have the following questions when I want to submit the forecasting result. Please kindly to help me.
1. Where can I find the submit data format in eval.ai as in can not see any detail instructions argoverse.github.io.
2. I want to participate the muti-agent motion forecasting challenge but I don't know which timestamp and agent should I concern, can anybody tell me the detail instructions? 

Many thanks.

soyons, 2023-10-15:
find solution

closed 2023-10-15 by soyons

cancui19, 2023-10-26:
@soyons Hi, would you mind kindly letting me know how you solve this issue? I cannot find the instructions either...

soyons, 2023-10-28:
> @soyons Hi, would you mind kindly letting me know how you solve this issue? I cannot find the instructions either...
I found the result from https://github.com/jchengai/forecast-mae/blob/main/src/model/multiagent/model_forecast_ma.py#L81
It may give you the answer.

cancui19, 2023-10-30:
@soyons Thanks for following up. One more question is that we cannot access the submission instructions (argoverse.github.io) for eval.ai. Could you show us the instructions for submitting our results? Thanks in advance.