argoverse/av2-api #237: Failed trying to run `generate_forecasting_scenario_visualizations.py` on python 3.11

SalamanderXing, 2023-11-30:
Hi, I downloaded the dataset and tried to run the script `generate_forecasting_scenario_visualizations.py`
This is how I'm running the script from the terminal:

```bash
python generate_forecasting_scenario_visualizations.py  --argoverse-scenario-dir=./train  --viz-output-dir=./video
```

This is the issue I got:
```
Traceback (most recent call last):
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/avi_scripts/generate_forecasting_scenario_visualizations.py", line 136, in <module>
    run_generate_scenario_visualizations()
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/avi_scripts/generate_forecasting_scenario_visualizations.py", line 126, in run_generate_scenario_visualizations
    generate_scenario_visualizations(
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/avi_scripts/generate_forecasting_scenario_visualizations.py", line 79, in generate_scenario_visualizations
    Parallel(n_jobs=1)(
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/joblib/parallel.py", line 1863, in __call__
    return output if self.return_generator else list(output)
                                                ^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/joblib/parallel.py", line 1792, in _get_sequential_output
    res = func(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/avi_scripts/generate_forecasting_scenario_visualizations.py", line 72, in generate_scenario_visualization
    visualize_scenario(scenario, static_map, viz_save_path)
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/av2/datasets/motion_forecasting/viz/scenario_visualization.py", line 67, in visualize_scenario
    cur_plot_bounds = _plot_actor_tracks(ax, scenario, timestep)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/av2/datasets/motion_forecasting/viz/scenario_visualization.py", line 172, in _plot_actor_tracks
    _plot_actor_bounding_box(
  File "/home/bluesk/Documents/mazen_thesis/mazen_thesis/mzth/lib/python3.11/site-packages/av2/datasets/motion_forecasting/viz/scenario_visualization.py", line 246, in _plot_actor_bounding_box
    vehicle_bounding_box = Rectangle(
                           ^^^^^^^^^^
TypeError: Rectangle.__init__() takes 4 positional arguments but 5 were given
```
This is the output of `pip freeze`:
```
argcomplete==3.1.6
av==11.0.0
av2==0.2.1
certifi==2023.11.17
click==8.1.7
colorlog==6.7.0
contourpy==1.2.0
cycler==0.12.1
distlib==0.3.7
filelock==3.13.1
fonttools==4.45.1
joblib==1.3.2
kiwisolver==1.4.5
llvmlite==0.41.1
markdown-it-py==3.0.0
matplotlib==3.8.2
mdurl==0.1.2
nox==2023.4.22
numba==0.58.1
numpy==1.26.2
opencv-python==4.8.1.78
packaging==23.2
pandas==2.1.3
Pillow==10.1.0
platformdirs==4.0.0
pyarrow==14.0.1
Pygments==2.17.2
pyparsing==3.1.1
pyproj==3.6.1
python-dateutil==2.8.2
pytz==2023.3.post1
rich==13.7.0
scipy==1.11.4
six==1.16.0
tzdata==2023.3
virtualenv==20.24.7
```
My python version is 3.11.6 and I'm running on x86_64 Linux 6.6.3-arch1-1

argoverse-admin, 2023-12-17:
Hi @SalamanderXing, thanks for raising the issue. We are going to go ahead and mark this as resolved now, since https://github.com/argoverse/av2-api/pull/235 has been merged (thanks for your PR in https://github.com/argoverse/av2-api/pull/238), due to the now required `angle` positional argument in the matplotlib call.

kobayashikona, 2024-02-05:
Hi, it seems that this issue still exists when I install av2 through conda directly.
My python version is 3.10.13, when I run the script `generate_forecasting_scenario_visualizations.py`, the terminal will feedback the following message:
```
File "/home/demo/anaconda3/envs/DenseTNT/lib/python3.10/site-packages/av2/datasets/motion_forecasting/viz/scenario_visualization.py", line 246, in _plot_actor_bounding_box
    vehicle_bounding_box = Rectangle(
TypeError: Rectangle.__init__() takes 4 positional arguments but 5 were given
```

rhythm-e, 2024-03-18:
I also come up with this issue, it can be solved by adding ''**angle=**'' in function 
```python
 vehicle_bounding_box = Rectangle(
          (pivot_x, pivot_y), bbox_length, bbox_width, angle=np.degrees(heading), color=color, zorder=_BOUNDING_BOX_ZORDER
      )
```
located in 
`.../python3.9/site-packages/av2/datasets/motion_forecasting/viz/scenario_visualization.py`