argoverse/av2-api #257: newest API 0.3.0 installation fails : ERROR: Could not build wheels for av2, which is required to install pyproject.toml-based projects

arbor03, 2024-03-27:
Hi, I downloaded the source code of the main branch for compilation, and an error occurred as shown follow:

 ```python
C:\Users\dell\Downloads\av2-api-main>pip install .
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Processing c:\users\dell\downloads\av2-api-main
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting av (from av2==0.3.0)
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/88/e3/effe4e9903d9b828ff76d527318568df7fe8d9bd2aaa6d147bdcd8cc3fbf/av-12.0.0-cp39-cp39-win_amd64.whl (26.4 MB)
Requirement already satisfied: click in c:\python\python39\lib\site-packages (from av2==0.3.0) (8.1.3)
Requirement already satisfied: joblib in c:\python\python39\lib\site-packages (from av2==0.3.0) (1.2.0)
Collecting kornia (from av2==0.3.0)
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/ac/fa/5612c4b1ad83b3044062e9dd0ca3c91937d8023cff0836269e18573655b0/kornia-0.7.2-py2.py3-none-any.whl (825 kB)
Requirement already satisfied: matplotlib in c:\users\dell\appdata\roaming\python\python39\site-packages (from av2==0.3.0) (3.6.1)
Requirement already satisfied: nox in c:\python\python39\lib\site-packages (from av2==0.3.0) (2024.3.2)
Requirement already satisfied: numba in c:\python\python39\lib\site-packages (from av2==0.3.0) (0.58.0)
Requirement already satisfied: numpy in c:\python\python39\lib\site-packages (from av2==0.3.0) (1.24.4)
Requirement already satisfied: opencv-python in c:\python\python39\lib\site-packages (from av2==0.3.0) (4.6.0.66)
Requirement already satisfied: pandas in c:\python\python39\lib\site-packages (from av2==0.3.0) (1.3.5)
Collecting polars (from av2==0.3.0)
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/ed/bd/35e3480f180c3e9bc57ef402c0b6e881da6909aaeff4eb31790e2e07beea/polars-0.20.16-cp38-abi3-win_amd64.whl (26.1 MB)
Requirement already satisfied: pyarrow in c:\python\python39\lib\site-packages (from av2==0.3.0) (15.0.2)
Requirement already satisfied: pyproj in c:\python\python39\lib\site-packages (from av2==0.3.0) (3.6.1)
Requirement already satisfied: rich in c:\python\python39\lib\site-packages (from av2==0.3.0) (13.5.3)
Requirement already satisfied: scipy in c:\python\python39\lib\site-packages (from av2==0.3.0) (1.11.3)
Requirement already satisfied: torch in c:\python\python39\lib\site-packages (from av2==0.3.0) (2.1.0+cu118)
Requirement already satisfied: tqdm in c:\python\python39\lib\site-packages (from av2==0.3.0) (4.64.1)
Collecting universal_pathlib (from av2==0.3.0)
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/63/a2/d8fe0f7e233c5e0201442d82279f3a0ffb30edcb8d7c4c5eba695c16647f/universal_pathlib-0.2.2-py3-none-any.whl (46 kB)
Requirement already satisfied: colorama in c:\python\python39\lib\site-packages (from click->av2==0.3.0) (0.4.5)
Collecting kornia-rs>=0.1.0 (from kornia->av2==0.3.0)
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/06/cf/1e1016c4b946c839a3cd9100ec4c67e55bddc85f10d921bb5857cff0e360/kornia_rs-0.1.2-cp39-none-win_amd64.whl (1.2 MB)
Requirement already satisfied: packaging in c:\python\python39\lib\site-packages (from kornia->av2==0.3.0) (23.2)
Requirement already satisfied: filelock in c:\python\python39\lib\site-packages (from torch->av2==0.3.0) (3.12.0)
Requirement already satisfied: typing-extensions in c:\python\python39\lib\site-packages (from torch->av2==0.3.0) (4.8.0)
Requirement already satisfied: sympy in c:\python\python39\lib\site-packages (from torch->av2==0.3.0) (1.12)
Requirement already satisfied: networkx in c:\python\python39\lib\site-packages (from torch->av2==0.3.0) (2.8.6)
Requirement already satisfied: jinja2 in c:\python\python39\lib\site-packages (from torch->av2==0.3.0) (3.1.2)
Requirement already satisfied: fsspec in c:\python\python39\lib\site-packages (from torch->av2==0.3.0) (2023.9.2)
Requirement already satisfied: contourpy>=1.0.1 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (1.0.5)
Requirement already satisfied: cycler>=0.10 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (4.37.3)
Requirement already satisfied: kiwisolver>=1.0.1 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (1.4.4)
Requirement already satisfied: pillow>=6.2.0 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (9.2.0)
Requirement already satisfied: pyparsing>=2.2.1 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (3.0.9)
Requirement already satisfied: python-dateutil>=2.7 in c:\python\python39\lib\site-packages (from matplotlib->av2==0.3.0) (2.8.2)
Requirement already satisfied: argcomplete<4.0,>=1.9.4 in c:\python\python39\lib\site-packages (from nox->av2==0.3.0) (3.2.3)
Requirement already satisfied: colorlog<7.0.0,>=2.6.1 in c:\python\python39\lib\site-packages (from nox->av2==0.3.0) (6.8.2)
Requirement already satisfied: virtualenv>=20.14.1 in c:\python\python39\lib\site-packages (from nox->av2==0.3.0) (20.23.0)
Requirement already satisfied: llvmlite<0.42,>=0.41.0dev0 in c:\python\python39\lib\site-packages (from numba->av2==0.3.0) (0.41.0)
Requirement already satisfied: pytz>=2017.3 in c:\python\python39\lib\site-packages (from pandas->av2==0.3.0) (2022.2.1)
Requirement already satisfied: certifi in c:\python\python39\lib\site-packages (from pyproj->av2==0.3.0) (2022.9.24)
Requirement already satisfied: markdown-it-py>=2.2.0 in c:\python\python39\lib\site-packages (from rich->av2==0.3.0) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in c:\python\python39\lib\site-packages (from rich->av2==0.3.0) (2.13.0)
Requirement already satisfied: mdurl~=0.1 in c:\python\python39\lib\site-packages (from markdown-it-py>=2.2.0->rich->av2==0.3.0) (0.1.2)
Requirement already satisfied: six>=1.5 in c:\python\python39\lib\site-packages (from python-dateutil>=2.7->matplotlib->av2==0.3.0) (1.16.0)
Requirement already satisfied: distlib<1,>=0.3.6 in c:\python\python39\lib\site-packages (from virtualenv>=20.14.1->nox->av2==0.3.0) (0.3.6)
Requirement already satisfied: platformdirs<4,>=3.2 in c:\python\python39\lib\site-packages (from virtualenv>=20.14.1->nox->av2==0.3.0) (3.5.1)
Requirement already satisfied: MarkupSafe>=2.0 in c:\python\python39\lib\site-packages (from jinja2->torch->av2==0.3.0) (2.1.1)
Requirement already satisfied: mpmath>=0.19 in c:\python\python39\lib\site-packages (from sympy->torch->av2==0.3.0) (1.3.0)
Building wheels for collected packages: av2
  Building wheel for av2 (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for av2 (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [48 lines of output]
      Running `maturin pep517 build-wheel -i C:\Python\Python39\python.exe --compatibility off`
[... 35 lines cut ...]
        --> C:\Users\dell\.cargo\registry\src\index.crates.io-6f17d22bba15001f\ahash-0.8.6\src/lib.rs:99:42
         |
      99 | #![cfg_attr(feature = "stdsimd", feature(stdsimd))]
         |                                          ^^^^^^^

      error: aborting due to 1 previous error

      For more information about this error, try `rustc --explain E0635`.

      Error: command ['maturin', 'pep517', 'build-wheel', '-i', 'C:\\Python\\Python39\\python.exe', '--compatibility', 'off'] returned non-zero exit status 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for av2
Failed to build av2
ERROR: Could not build wheels for av2, which is required to install pyproject.toml-based projects
```

How can I solve this?

I would like to use the latest version of the API. 
Are there plans to upload it to PyPi soon? 
Or you could create a wheel package to facilitate installation for researchers?

Best Regards

Carrotsniper, 2024-03-28:
I have the same issues, play help us to install it.

WHuMaverickCN, 2024-04-07:
I got the same question one time. How can I fix it plz.

Carrotsniper, 2024-04-07:
> I got the same question one time. How can I fix it plz.

I have solved this problem following this issue: https://github.com/argoverse/av2-api/issues/211
Go ahead and check it.

Cheers

WHuMaverickCN, 2024-04-07:
> > I got the same question one time. How can I fix it plz.
> 
> I have solved this problem following this issue: #211 Go ahead and check it.
> 
> Cheers

Got it. 谢谢hxd