argoverse/av2-api #260: Submission blocked on Argoverse 2

kingwmk, 2024-04-17:
Hello. I have a submission blocked problem on Argoverse 2: Motion Forecasting Competition's Multi-world prediction channel. 
The status always is 'submitted'.
Thank you for your help. https://eval.ai/web/challenges/challenge-page/1719/my-submission

wqi, 2024-04-18:
Hi @kingwmk, thanks for reaching out!

The EvalAI has since been resolved - please feel free to re-open this issue if the problem persists.

closed 2024-04-18 by wqi

kingwmk, 2024-04-21:
Thank you so much for fixing this issue. However, I get a new problem that I can not upload the submission.parquet file. It blocked in the uploading process though I changed many networks.
[image]