argoverse/av2-api #282: Submission Problem for motion forecasting

skylineo, 2024-10-13:
When submitting results in the argoverse motion forecasting challenge, the previous attempts have been successful, but the results of the last two submissions have been stuck in the "running" stage. How can I solve this? The submission page is: https://eval.ai/web/challenges/challenge-page/454/my-submission
Can you reply as soon as possible? Thank you a lot!

wqi, 2025-03-24:
Should be back online again, thanks for your patience!

I've requeued most of the stuck submissions, but if your submission doesn't show any results, please just submit it again.

closed 2025-03-24 by wqi

HHHzp, 2025-04-08:
> When submitting results in the argoverse motion forecasting challenge, the previous attempts have been successful, but the results of the last two submissions have been stuck in the "running" stage. How can I solve this? The submission page is: https://eval.ai/web/challenges/challenge-page/454/my-submission在 argorse 运动预测挑战赛中提交结果时，之前的尝试都取得了成功，但最近两次提交的结果却一直停留在“运行”阶段。我该怎么解决这个问题？提交页面为：https://eval.ai/web/challenges/challenge-page/454/my-submission Can you reply as soon as possible? Thank you a lot!你能尽快回复吗？非常感谢！

I have also encountered this situation and have been stuck in the submission stage. I don't know why, but I remember it was okay last week

HHHzp, 2025-04-08:
> Should be back online again, thanks for your patience!应该会重新上线，感谢您的耐心等待！
> 
> I've requeued most of the stuck submissions, but if your submission doesn't show any results, please just submit it again.我已经重新排列了大多数卡住的提交，但如果您的提交没有显示任何结果，请重新提交。

I have also experienced this situation these days and have been stuck in the submission stage. I don't know why, but I remember it was okay last week. I have also resubmitted many times but it still doesn't work. How can I solve it

wqi, 2025-04-08:
Should be back online now!

HHHzp, 2025-04-09:
> Should be back online now!

ok,thanks,let me try again!

HHHzp, 2025-04-09:
It seems to have been stuck in the submitted status for almost an hour now. @wqi 

[image]