argoverse/av2-api #287: Submission cannot run

Yurriki, 2024-12-10:
@wqi Sorry for borthering you,however, my submission  on EvalAI cannot run and has been in a "submitted" state for a long time.  Is the server updating or is there any other issue?
Best wishes!

wqi, 2024-12-11:
Should be back online now, thanks for your patience!

closed 2024-12-11 by wqi

XinGP, 2024-12-13:
@wqi Hello, I have also encountered a similar problem. When I was uploading the parcel file, the webpage got stuck and there was no response for a long time.

XinGP, 2024-12-13:
Hello, I have also encountered a similar problem. When I was uploading the parquet file, the webpage got stuck and there was no response for a long time.

wqi, 2024-12-13:
When were these submissions created? I checked the evaluation server and it doesn't seem like anything's currently stuck in queue.

XinGP, 2024-12-13:
The submission was made just 10 minutes ago, and my ID is “whut-its
[image]
”

XinGP, 2024-12-13:
There are some issues with my description. I have been stuck here for almost an hour and decided to consult 10 minutes ago

wqi, 2024-12-13:
This appears to be a problem with the EvalAI platform unfortunately. I checked the submission queue and it doesn't seem like an entry is being created for the evaluation server to run on.

I do recall a similar issue being raised about a year ago, which ended up being traced back to some kind of networking issue. Could you try 1) submitting using a different machine on a different network or 2) using the CLI?

XinGP, 2024-12-17:
Sorry to bother you again, we found that uploading can be done normally on argoverse2 single, but Muti World has changed many machines and networks, including using CLI, and still cannot upload. Could you please check if your machine can upload normally?

XinGP, 2024-12-18:
> @wqi Sorry to bother you again, we found that uploading can be done normally on argoverse2 single, but Muti World has changed many machines and networks, including using CLI, and still cannot upload. Could you please check if your machine can upload normally?

@wqi Thank you.

wqi, 2024-12-18:
I was able to successfully upload and evaluate a parquet submission to the multi-world challenge, so it doesn't appear to be an issue with the evaluation server itself.

How large is your submission file? Anecdotally, it does seem like uploading large files to EvalAI can take a while.

XinGP, 2024-12-18:
> I was able to successfully upload and evaluate a parquet submission to the multi-world challenge, so it doesn't appear to be an issue with the evaluation server itself.
> 
> How large is your submission file? Anecdotally, it does seem like uploading large files to EvalAI can take a while.

@wqi About 500mb, can you tell me if you uploaded it through web or CLI?

XinGP, 2024-12-18:
> I was able to successfully upload and evaluate a parquet submission to the multi-world challenge, so it doesn't appear to be an issue with the evaluation server itself.
> 
> How large is your submission file? Anecdotally, it does seem like uploading large files to EvalAI can take a while.

[image]
It seems that the reason is that the file is too large

XinGP, 2024-12-19:
@wqi Thank you for your multiple replies. This issue has been resolved. I can submit normally using CLI in Windows. CLI under Linux often reports errors and it is uncertain whether it is a bug.