argoverse/av2-api #304: Argoverse2 motion forecasting competition leaderboard evaluation Failed

Morphlingxxx, 2025-03-24:
The argoverse2 motion forecasting competition leaderboard often cannot be uploaded. It has been down since the day before yesterday.

wqi, 2025-03-24:
Should be back online again, thanks for your patience!

I've requeued most of the stuck submissions, but if your submission doesn't show any results, please just submit it again.

MKnoche, 2025-03-26:
I think this issue still persist, all of my submissions fail with `KeyError("No predictions found for scenario ('0000b329-f890-4c2b-93f2-7e2413d4ca5b',).")`, although the scenario is in the parquet file I have uploaded. I also tried some old files which worked in the past and they fail as well.

wqi, 2025-03-26:
Hi @MKnoche, I can confirm that I can reproduce this issue on the multi-agen

wqi, 2025-03-26:
Should be fixed now. I reprocessed your latest submission for you.

Feel free to re-open the issue if any issues persist. :-)

closed 2025-03-26 by wqi