argoverse/av2-api #318: Inquiry About Reopening of Argoverse2 2025 Challenge Evaluation Server

gnillling, 2025-06-16:
I noticed that the Argoverse2 2025 Challenges officially ended on June 8th. However, I am currently conducting experiments and would like to evaluate my model's performance using the official evaluation protocol. Could you please let me know if there are any plans to reopen the evaluation server, or if there is an alternative way to submit results for benchmarking?

penglo, 2025-06-19:
> I noticed that the Argoverse2 2025 Challenges officially ended on June 8th. However, I am currently conducting experiments and would like to evaluate my model's performance using the official [...]

Hello, I’ve encountered the same issue you mentioned. Since the evaluation server is now closed, I’ve also been looking for alternative ways to evaluate my results. I was wondering if you’ve found a solution? If so, would you be willing to share your approach so I could learn from it? I’d really appreciate it!

penglo, 2025-06-22:
> Which challenge are you interested in evaluating?

argoverse2，single agent challenge ，thank you very much

neeharperi, 2025-06-22:
@wqi Should be able to help you

NoahWangc, 2025-06-23:
> Which challenge are you interested in evaluating?

I have the same question, and I hope to open "Argoverse 2: Motion Forecasting Competition" again as soon as possible. Thank you very much!

wqi, 2025-06-23:
Hi @penglo and @NoahWangc , thanks for your continued interest! Both the single-agent and multi-world forecasting evaluation servers have been reopened for submission.

Please let me know if you experience any issues with evaluation.

closed 2025-06-23 by wqi

Lukas88664, 2025-11-08:
@wqi could you please reopen argoverse2，single agent challenge？